adafruit/Adafruit_INA219 #20: Wire.setClock(highspeed) implementation unclear

bc, 2018-08-27:
Hi,
I'm working on a robotic experiment and want high speedy watt measurement rate. I'm using the Adafruit INA291 with no resistor modifications. I'd like to speed up the I2C to the fastest rate possible, which is 3.4MhZ from the datasheet (https://cdn-shop.adafruit.com/datasheets/ina219.pdf), or at least the fastest the Arduino Uno can handle. I see that this Adafruit_INA219 library uses the Wire package to set up the connection to the INA219, but setting Wire.setClock(3400000); does not appear to increase the rate. I get ~500s/s when printing to Serial when 1000 samples have been observed (via hand timing of 10000 samples).

How can I increase the I2C rate for the INA219 from Arduino? Is there a function/parameter I can call or pass?

- Arduino board: Arduino Uno

- Arduino IDE version 1.8.5

- Use case: Adding watts over time for a variety of loads on a power supply. I will not be printing incrementally to Serial.

Thanks so much!

Brian

```#include <Wire.h>
#include <Adafruit_INA219.h>

Adafruit_INA219 ina219;
int counter;
float power_mW;
void setup(void) 
{
  pinMode(12, INPUT_PULLUP); 
  Serial.begin(115200);
  Wire.setClock(3400000);
  while (!Serial) {
      // will pause Zero, Leonardo, etc until serial console opens
      delay(1);
  }
  uint32_t currentFrequency;
  ina219.begin();
  power_mW = 0;
}

void loop(void) 
{
  if (counter % 1000 == 0){
    Serial.println(counter);
    }
  power_mW = ina219.getPower_mW();
  counter += 1;
}```

trycoon, 2018-11-10:
I don't know if this delay is needed or not, try comment it out: https://github.com/adafruit/Adafruit_INA219/blob/master/Adafruit_INA219.cpp#L73

ladyada, 2019-06-06:
hiya - closing cause no updates! re-open if you have other q's

M-Hellemans, 2021-01-19:
Better late than never and google brought me here so thought I'd share for future reference. 

This works: 
`
#include <Adafruit_INA219.h>
#include <Wire.h>

Adafruit_INA219 ina219;

ina219.begin(&Wire);
Wire.setClock(3400000);
`

The important part is to pass Wire by reference, and changing the clock **afterwards**. .begin() seems to reset the clockspeed.

closed 2021-11-24 by caternuson