adafruit/Adafruit_INA219 #29: Very low shuntvoltage

Mcpixel53, 2019-04-02:
Hello,

Any idea what can be causing a very low shunt (and so, load) voltage?

I have a 4W 3.2Ohm resistance in the negative VIN, the current meassurement seems to be correct, around 800mA for Lion battery testing.

Thanks in advance for any answer!

ladyada, 2019-04-02:
hiya, this github repo is not for hardware design support - thanks :)

closed 2019-04-02 by ladyada

Mcpixel53, 2019-04-02:
I think you misunderstood me, I'm using this library and didn't touch the module at all. 

It looks pretty straight forward so I thought It could be a firmware issue, I only showed my "hardware" configuration to prove I'm using it right.

Could U tell me if I need to make any special calibration for INA219-GY219 or it is supposed to work with the provided example?

Do U think It may be a hardware issue?   You're welcome =)

ladyada, 2019-04-02:
hiya if you are using a GY219 module please contact the vendor - they can provide tech support :)

Mcpixel53, 2019-04-02:
The fix was simply ground the negative with the arduino (\*self-facepalm\*). 
In case it is useful for others wondering the same:
https://cdn-learn.adafruit.com/downloads/pdf/adafruit-ina219-current-sensor-
[image]