adafruit/Adafruit_INA219 #41: busVoltage should be handled as unsigned int

andreobi, 2020-06-07:
Error: measured voltages over 16.3V are returned as negative value. 
Problem: the i2c value is treaded as signed INT 
Proposal: change it to unsigned INT as done below and the .h file accordingly.

Andre

uint16_t INA219::busVoltageRaw() {
  _bus_voltage_register = read16(V_BUS_R);
  _overflow = bitRead(_bus_voltage_register, OVF_B);     // overflow bit
  _ready    = bitRead(_bus_voltage_register, CNVR_B);    // ready bit
  return _bus_voltage_register;
}

float INA219::busVoltage() {
  uint16_t temp;
  temp = busVoltageRaw();
  temp >>= 3;
  return (temp * 0.004);
}

ladyada, 2020-06-07:
thanks, please submit a PR :)

andreobi, 2020-06-07:
I used an older version. It's already fixed.

closed 2020-06-07 by andreobi