adafruit/Adafruit_INA219 #42: Adafruit_BusIO_Register.h requirement

vitormhenrique, 2020-08-01:
library imports Adafruit_BusIO_Register.h that seems no where to be found?!

vitormhenrique, 2020-08-01:
I guess update documentation to have https://github.com/adafruit/Adafruit_BusIO as a requirement.

a lot of people this days don't use Arduino library management, so just having it documented it would be good! 👍

caternuson, 2020-08-02:
In a way, it is documented:
https://github.com/adafruit/Adafruit_INA219/blob/master/library.properties#L10

What is being used other than the Arduino Library Manager?

vitormhenrique, 2020-08-02:
hmm...

personally I have been using and seen an increase of platform io.... (sure, that still use library manager, but I prefer keeping my libraries on the project folder)

Anyway, thanks!

closed 2020-08-02 by ladyada

ladyada, 2020-08-03:
ok, thanks! tagging @ivankravets since it may be a platformio question