adafruit/Adafruit_INA219 #49: Confusion between "load voltage" and "bus voltage".

Koepel, 2021-04-02:
There is confusion between "load voltage" and "bus voltage" that is used in the library.
I made this issue because of this forum topic: [https://forum.arduino.cc/index.php?topic=735283.0](url).

Example:  
When a 12V car battery provides power to the devices in a car and a rear window heater is the load, then the 12V from the car battery is the "bus voltage" and the voltage over the heater is the "load voltage".  
The 12V is connected to VIN+ and the heater is connected to VIN-.

It has always been that way, even on the website of Texas Instruments.

The datasheet of the INA219 is confusing. They say that the bus voltage is measured at pin VIN-  
That is not okay. The Adafruit library went along that route and ended up exchanging the names "load voltage" and "bus voltage". So now the confusion is complete (and wrong).

I suggest to make new names, and add explanation every time the old names are used.

Aldo-aldo, 2024-08-19:
I have the same problem

DashApple, 2026-05-04:
I had noticed this aswell.

The INA219 Reads the voltage from the Shunt - side to ground so this would be the load voltage and this is what the INA219 messures and shows this from. 

The Bus voltage is the voltage that supplies the load from the + side of the shunt so to work this out VBus = Vload + Shunt. It seemed the libary had this backwards ; loadvoltage = busvoltage + (shuntvoltage / 1000); This will give a wrong value as the load voltage is now reported as more than the supply.

Since the libary abd the INA219 datasheet lables the votage it reads as bus voltage this makes the confusion worse as this isnt what is messured, Its actually the load voltage in this context