adafruit/Adafruit_INA219 #61: Multiple sensors

Krki012, 2025-01-29:
I have a project in which I need two INA219 sensors. Both are used to stop small DC motors when overload occurs. 

My question now is: How do I specify which sensor is on which address?

I have tried many things but they always result in error and I just can't get them both to work at the same time. 

Can anyone help me please?

caternuson, 2025-01-29:
> How do I specify which sensor is on which address?

Create an instance for each sensor and specify the I2C address in the constructor, ex:
```cpp
Adafruit_INA219 sensor1;        // uses address 0x40 (default)
Adafruit_INA219 sensor2(0x41);  // uses address 0x41
```

caternuson, 2025-01-29:
This is related to #59

Krki012, 2025-01-29:
Aaaa like that ...
I've been trying like Adafruit_INA219(0x41) sensor; 
I'm dumb XD
Thank you

closed 2025-01-29 by ladyada